lewisxhe/XPowersLib #3: AXP192: getBattVoltage() always returns 0

cyberman54, 2022-09-17:
AXP192_showstatus(): [src/power.cpp] Battery charging, **0.00V** @ 752mAh

code:
```
if (pmu.isBatteryConnect())
    if (pmu.isCharging())
      ESP_LOGI(TAG, "Battery charging, %.2fV @ %.0fmAh",
               pmu.getBattVoltage() / 1000, pmu.getBatteryChargeCurrent());
```

lewisxhe, 2022-09-18:
Has the battery voltage detection function been enabled? I'm fine here.
And you don't need to judge whether the battery is connected or not. It has been judged whether the battery is connected in getBattVoltage and getBatteryPercent. If there is no battery connection, getBattVoltage will return 0, and getBatteryPercent will return -1.

Below is my test code:

```
#define XPOWERS_CHIP_AXP192

#include <Wire.h>
#include <Arduino.h>
#include "XPowersLib.h"

bool  pmu_flag = 0;
XPowersPMU PMU;

const uint8_t i2c_sda = 21;
const uint8_t i2c_scl = 22;
const uint8_t pmu_irq_pin = 35;

void setFlag(void)
{
    pmu_flag = true;
}

void setup()
{
    Serial.begin(115200);

    bool result = PMU.begin(Wire, AXP192_SLAVE_ADDRESS, i2c_sda, i2c_scl);

    if (result == false) {
        Serial.println("PMU is not online..."); while (1)delay(50);
    }

    // It is necessary to disable the detection function of the TS pin on the board
    // without the battery temperature detection function, otherwise it will cause abnormal charging
    PMU.disableTSPinMeasure();

    // Enable internal ADC detection
    PMU.enableBattDetection();
    PMU.enableVbusVoltageMeasure();
    PMU.enableBattVoltageMeasure();
    PMU.enableSystemVoltageMeasure();

    while (1) {
        Serial.print("isCharging:"); Serial.println(PMU.isCharging() ? "YES" : "NO");
        Serial.print("isDischarge:"); Serial.println(PMU.isDischarge() ? "YES" : "NO");
        Serial.print("isVbusIn:"); Serial.println(PMU.isVbusIn() ? "YES" : "NO");
        Serial.print("getBattVoltage:"); Serial.print(PMU.getBattVoltage()); Serial.println("mV");
        Serial.print("getVbusVoltage:"); Serial.print(PMU.getVbusVoltage()); Serial.println("mV");
        Serial.print("getSystemVoltage:"); Serial.print(PMU.getSystemVoltage()); Serial.println("mV");
        Serial.print("getTemperature:"); Serial.print(PMU.getTemperature()); Serial.println("*C");
        Serial.print("getBatteryPercent:"); Serial.print(PMU.getBatteryPercent()); Serial.println("%");
        Serial.println();
        delay(2000);
    }
}

void loop()
{
}

```

cyberman54, 2022-09-18:
Turned out, this was a display masking error.

```
ESP_LOGI(TAG, "Battery charging, %.2fV @ %.0fmAh",
               pmu.getBattVoltage() / 1000.0, pmu.getBatteryChargeCurrent());
```

works (note the 1000.0 instead of 1000)

(With the old AXP192 lib this worked, maybe there was a return type change for getBattVoltage?)

closed 2022-09-18 by cyberman54