lewisxhe/XPowersLib #14: PMU is not online...Please help me!

khoa21214475, 2023-10-16:
Dear lewisxhe,
when I scan i2c address, I received 0x34.
but when I use example AXP2101_Voltage_Example, I received "PMU is not online"
I don't know why, i2c bus is ok, 2 resistors pullup is ok, but my device can't connect to AXP2101!
//////////////////////////////////////////////////////////////////////////////////////////////////////////////
Here is my code:

#define XPOWERS_CHIP_AXP2101

#include <Wire.h>
#include <Arduino.h>
#include "XPowersLib.h"

XPowersPMU PMU;

#ifndef CONFIG_PMU_SDA
#define CONFIG_PMU_SDA 10
#endif

#ifndef CONFIG_PMU_SCL
#define CONFIG_PMU_SCL 9
#endif

#ifndef CONFIG_PMU_IRQ
#define CONFIG_PMU_IRQ 8
#endif

const uint8_t i2c_sda = CONFIG_PMU_SDA;
const uint8_t i2c_scl = CONFIG_PMU_SCL;
const uint8_t pmu_irq_pin = CONFIG_PMU_IRQ;

uint16_t targetVol;
//uint16_t vol = 0;

void setup()
{
    Serial.begin(115200);
    Wire.begin();
    bool result = PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, i2c_sda, i2c_scl);
    delay(100);

    if (result == false) {
      Serial.println("PMU is not online..."); while (1)delay(50);
    }

    Serial.println("AXP2101 Power Output Test.");

    PMU.disableDC2();
    PMU.disableDC3();
    PMU.disableDC4();
    PMU.disableDC5();
    PMU.disableALDO1();
    PMU.disableALDO2();
    PMU.disableALDO3();
    PMU.disableALDO4();
    PMU.disableBLDO1();
    PMU.disableBLDO2();
    PMU.disableCPUSLDO();
    PMU.disableDLDO1();
    PMU.disableDLDO2();

    //DCDC1
    PMU.setDC1Voltage(3300);
    
    //DCDC3
    PMU.setDC3Voltage(3300);
    delay(1);
    targetVol = PMU.getDC3Voltage();
    
    //ALDO1 IMAX=300mA
    //500~3500mV, 100mV/step,31steps
    PMU.setALDO1Voltage(3300);
    delay(1);
    targetVol = PMU.getALDO1Voltage();
        
    //ALDO2 IMAX=300mA
    //500~3500mV, 100mV/step,31steps
    PMU.setALDO2Voltage(3300);
    delay(1);
    targetVol = PMU.getALDO2Voltage();
    
    //ALDO3 IMAX=300mA
    //500~3500mV, 100mV/step,31steps

    PMU.setALDO3Voltage(3300);
    delay(1);
    targetVol = PMU.getALDO3Voltage();
    
    //ALDO4 IMAX=300mA
    //500~3500mV, 100mV/step,31steps
    PMU.setALDO4Voltage(3300);
    delay(1);
    targetVol = PMU.getALDO4Voltage();
    
    //BLDO1 IMAX=300mA
    //500~3500mV, 100mV/step,31steps
    PMU.setBLDO1Voltage(3300);
    delay(1);
    targetVol = PMU.getBLDO1Voltage();
    
    PMU.enableDC1();
    PMU.enableDC3();

    PMU.enableALDO1();
    PMU.enableALDO2();
    PMU.enableALDO3();
    PMU.enableALDO4();
    PMU.enableBLDO1();
}
/////////////////////////////////////////////////////////////
void loop()
{
    delay(100);
}

lewisxhe, 2023-10-17:
Change to the following code to test, clearly indicating which pin I2C SDA, SCL is connected to.
And check if the ID is 0x03

```
#define XPOWERS_CHIP_AXP2101

#include <Wire.h>
#include <Arduino.h>
#include "XPowersLib.h"

XPowersPMU PMU;

const uint8_t i2c_sda = You Board SDA;
const uint8_t i2c_scl = You Board SCL;

void setup()
{
    Serial.begin(115200);
    //Wire.begin() ; Don't use Wire.begin()unless you are sure you are selecting the correct i2c pin
    bool result = PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, i2c_sda, i2c_scl);
    delay(100);

    Serial.printf("getID:0x%x\n", PMU.getChipID());

    if (result == false) {
      Serial.println("PMU is not online..."); while (1)delay(50);
    }

}

void loop(){
}
```

khoa21214475, 2023-10-17:
[image]
Dear lewisxhe,

Im tested, but the result is 0xFF

lewisxhe, 2023-10-18:
Can you search for the device with 0X34 and get the ID as 0XFF?
Which board are you using? Did you buy the chip yourself?

khoa21214475, 2023-10-18:
Thanks Lewisxhe, I will try to find out this problem by your way!
I use STM32L432 chip, I bought AXP2101 via taobao.com (link: https://item.taobao.com/item.htm?spm=a21n57.1.0.0.3682523cCuZCAO&id=730904931624&ns=1&abbucket=8#detail)
Many thank Lewisxhe!

khoa21214475, 2023-10-18:
I read register at address 0x03 and the result is 0x74

khoa21214475, 2023-10-18:
I read register at address 0x03 and the result is 0x74 (not same at value 0x4A in this library)

khoa21214475, 2023-10-18:
I read register at address 0x03 and the result is 0x4A.
May be this library can not use for STM32 duino. 
My code:

#include <Wire.h>
const int deviceAddress = 0x34;  // Địa chỉ của thiết bị I2C (thay thế bằng địa chỉ thực tế)
void setup() {
  Serial.begin(115200);
  Wire.begin();
}
void loop() {
  // Địa chỉ thanh ghi bạn muốn đọc
  int registerAddress = 0x03;  // Thay thế bằng địa chỉ thanh ghi thực tế

  // Bắt đầu gửi yêu cầu đến thiết bị
  Wire.beginTransmission(deviceAddress);
  Wire.write(registerAddress);
  Wire.endTransmission();

  // Đọc 1 byte từ thiết bị
  Wire.requestFrom(deviceAddress, 1);
  if (Wire.available()) {
    int data = Wire.read();  // Đọc dữ liệu từ thanh ghi
    Serial.print("register 0x");
    Serial.print(registerAddress, HEX);
    Serial.print(": 0x");
    Serial.println(data,HEX);
  }

  delay(1000);  // Chờ 1 giây trước khi đọc lại
}

lewisxhe, 2023-10-19:
Okay, I passed the test using Nucleo f411re

khoa21214475, 2023-10-20:
Really thank you very much. Your new update is running fine with STM32.
I sincerely thank you very much!

lyusupov, 2023-10-20:
> Your new update is running fine with STM32.

If you are happy with the fix made by Lewis - why don't you close this ticket ?

closed 2023-10-20 by khoa21214475