lewisxhe/XPowersLib #15: Equivalent of AXP202 library

NDC-SC, 2023-10-31:
Is there an equivalent library for a battery powered ESP32-S3 to the AXP202 library for the TTGO T-Watch.
Currant usage could be measured without a device, such as an INA219 module.
The getBattChargeCurrent() and getBattDischargeCurrent() function are very useful.

lewisxhe, 2023-11-16:
What you should be talking about is AXP2101, which does not have a discharge current measurement function. Isn’t there a lot of libraries available on IN219 github?

closed 2023-11-16 by lewisxhe

NDC-SC, 2023-11-16:
The advantage of the power management library for the TTGO T-Watch is that an external INA219 module is not required.
Is there a comparable power management library for the ESP32-S3 ?