lewisxhe/XPowersLib #20: Next release ?

mathieucarbou, 2024-03-15:
Would it be possible to release the new changes and `pio pkg publish` them ?
Thanks!

lewisxhe, 2024-03-15:
Of course, but wouldn't esp32 also be indexed into espressif?

mathieucarbou, 2024-03-15:
No for ESPIDF projects this is yet another repo. I was more asking of a new tag and new deployment in platformio repo: https://registry.platformio.org/libraries/lewisxhe/XPowersLib

lewisxhe, 2024-03-15:
Okay, it has been published. It will be indexed in a moment.

lewisxhe, 2024-03-15:
[image]

Done .

closed 2024-03-15 by lewisxhe

mathieucarbou, 2024-03-15:
Thanks a lot :-)
FYI I m working on this project which makes use of the new lilyfo boards and XPower lib and your TinyGSM fork: https://beelance.carbou.me

lewisxhe, 2024-03-15:
Cool project, thanks for sharing