lewisxhe/XPowersLib #21: Small charge current in constant charge(CC)! Please help me!

khoa21214475, 2024-04-04:
20:48:50.186 -> AXP2101 Started Successful.
20:48:50.186 -> DCDC1  :ENABLE   Voltage:3100 mV 
20:48:50.218 -> DCDC3  :DISABLE   Voltage:3300 mV 
20:48:50.218 -> ALDO1  :DISABLE   Voltage:3300 mV 
20:48:50.218 -> ALDO2  :DISABLE   Voltage:3100 mV 
20:48:50.218 -> ALDO3  :DISABLE   Voltage:3100 mV 
20:48:50.218 -> ALDO4  :DISABLE   Voltage:3100 mV 
20:48:50.218 -> BLDO1  :DISABLE   Voltage:3300 mV 
20:48:50.218 -> ->  getSysPowerDownVoltage:2600
20:48:50.218 -> DISABLE - HEX:0xffffffff 
20:48:50.218 -> ENABLE - HEX:0x18fcc0 
20:48:50.259 -> Setting Charge Target Current : 1000
20:48:50.259 -> Setting Charge Target Voltage : 4200
20:48:50.259 -> 
20:48:55.248 -> ---------------------------------------------------------------------------------------------------------
20:48:55.248 -> Satus1  Satus2  CHARG   DISC   STBY    VBUSIN    VGOOD    VBAT   VBUS   VSYS   Percentage    CHG_STATUS
20:48:55.248 -> (Bin)   (Bin)   (bool)  (bool) (bool)  (bool)    (bool)   (mV)   (mV)   (mV)      (%)           (str)  
20:48:55.248 -> ---------------------------------------------------------------------------------------------------------
20:48:55.294 -> 0b11000	0b10010	YES	NO 	NO 	YES	YES	3774	4771	4765	49	constant charge(CC)

20:49:06.297 -> =====================================================================
20:49:06.297 -> BatVoltage: 3772(49%)
20:49:06.297 -> >>>>>>register 0x0: 0b111000
20:49:06.297 -> >>>>>>register 0x1: 0b110010
20:49:06.297 -> >>>>>>register 0x67: 0b11100110
20:49:06.340 -> charge_status: 0b10

The charging current is only about 3-5mA, so the charging time exceeds 12H (set by reg67H[5:4]), and thus also interrupts the charging process. I can't fix this problem, please help me!

khoa21214475, 2024-04-11:
Please help me!

lewisxhe, 2024-04-11:
You want to set the charge current, but it doesn't change after setting it to 1000mA?

khoa21214475, 2024-04-12:
Yes, I set charge current to 1000mA:
    // Set constant current charge current limit
    if (!PMU.setChargerConstantCurr(XPOWERS_AXP2101_CHG_CUR_1000MA)) {
        Serial.println("Setting Charger Constant Current Failed!");
    }
And Charger Status is Constant Current (CC)
But the charging current is only from 3mA to 5 mA, so the charging time exceeds 12 hours so an error is reported through the charger status LED (1Hz blink) and stops charging.
Can you help me solve this problem, thank you very much!

lewisxhe, 2024-04-12:
Which board do you use? Looks like a hardware issue

khoa21214475, 2024-04-12:
[Schematic_BBKB_LORA_AXP2101_v1.0_28.10.2023_2024-04-12.pdf](https://github.com/lewisxhe/XPowersLib/files/14955193/Schematic_BBKB_LORA_AXP2101_v1.0_28.10.2023_2024-04-12.pdf)
It is my board, use stm32l432 and AXP2101

lewisxhe, 2024-04-12:
This problem has nothing to do with this library. It can be determined to be a hardware problem. Please ask your supplier about the charging method of the AXP2101 you are using. AXP2101 has two charging methods, and the hardware wiring methods are different.