lewisxhe/XPowersLib #23: Consider replacing the variable name PMU  in examples with anything else

hybrid-orbital, 2024-07-05:
Espressif defines a PMU variable declared as extern for some of their SOCs (in my case it was the ESP32H2): 
https://github.com/espressif/esp-idf/blob/1cb4f3468d64c9e8bde196b15df37cbcd4b31395/components/soc/esp32h2/include/soc/pmu_struct.h#L761
This code is precompiled into the esp32-arduino-lib library installed by Arduino. If you run the example programs on these SOCs, everything will get messed up due to the global PMU variable.

I encountered this problem when debugging the sleep function of ESP32-H2. It took me some time to troubleshoot this issue because the compiler doesn't directly prompt the error. 
The sleep-related functions of ESP32-H2 will reference the PMU variable, which will cause the sleep function to not run properly and cause core panic. 
https://github.com/espressif/esp-idf/blob/1cb4f3468d64c9e8bde196b15df37cbcd4b31395/components/esp_hw_support/sleep_wake_stub.c#L71-L74
https://github.com/espressif/esp-idf/blob/1cb4f3468d64c9e8bde196b15df37cbcd4b31395/components/esp_hw_support/sleep_modes.c#L2060

This problem can be reproduced by running AXP2101_Sleep_Example.ino on ESP32-H2. Switch 'Core Debug Level' to 'Verbose' and it should show something like
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
MEPC    : 0x40803bd8  RA      : 0x408058c8  SP      : 0x40817c50  GP      : 0x4080e194  
TP      : 0x40806dd8  T0      : 0x40003128  T1      : 0x40000038  T2      : 0x00000000  
S0/FP   : 0x80000000  S1      : 0x00000010  A0      : 0x4080e61c  A1      : 0x80000010  
A2      : 0x00000000  A3      : 0x40000000  A4      : 0x4080f090  A5      : 0x00000000  
A6      : 0x00000003  A7      : 0x00000037  S2      : 0x00000001  S3      : 0x00000000  
S4      : 0x004c494c  S5      : 0x00000001  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000001  S10     : 0x4080f74c  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000018  MTVAL   : 0x00008082  
MHARTID : 0x00000000  
```
and reboot

lewisxhe, 2024-07-08:
I am not using the AXP2101 chip on the ESP32H2, thanks for your report, I will make the change

lewisxhe, 2024-07-08:
Thanks for reminding me, the changes have been completed.

closed 2024-07-08 by lewisxhe