lewisxhe/XPowersLib #24: Temperature function for AXP 2101

LakinduBabySoap, 2024-08-08:
Hello, firstly thank you for creating and managing this library. It has been very helpful to track these parameters of the microcontrollers.

I have 2 T-Beams, one with the AXP192 and one with AXP 2101. My aim is to retrieve if the T-Beam is charging, temperature, battery voltage and percentage and forward it to TTN. For the AXP192, I can see all these parameters but for AXP2101, the example does not show how to retrieve temperature data.

However, when I add the statement ```PMU.getBattVoltage()``` and print it out and send it to ttn, it produces a faulty value. For example. it returns temperature as 58.2 degrees when it should be in the 30s range. Is this function not configured for the AXP2101?

Thank you.

lewisxhe, 2024-08-08:
See example [AXP2101_ADC_Example](https://github.com/lewisxhe/XPowersLib/blob/master/examples/AXP2101_ADC_Example/AXP2101_ADC_Example.ino)

LakinduBabySoap, 2024-08-09:
> See example [AXP2101_ADC_Example](https://github.com/lewisxhe/XPowersLib/blob/master/examples/AXP2101_ADC_Example/AXP2101_ADC_Example.ino)

Thank you for the reply. Unfortunately, when I run the program, it produces this error on loop.
```
invalid header: 0x61259440
invalid he�ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x61259440
invalid header: 0x61259440
```

I have run the ADC 2101 Example without any issue. Would appreciate if you can check this? I am not sure if it is a problem with the device.

Thank you.

lewisxhe, 2024-08-10:
Of course it is normal. The error is caused by selecting the wrong option in Arduino IDE. This example is for reference of another [esp32S3](https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G/issues/27) . Everything is normal.
[image]

LakinduBabySoap, 2024-08-15:
> Of course it is normal. The error is caused by selecting the wrong option in Arduino IDE. This example is for reference of another [...]

Apologies for the late reply. I changed the pin configurations of the code and now it works perfectly fine. You can close the issue.

Thanks!

closed 2024-08-15 by lewisxhe