lewisxhe/XPowersLib #31: Template class between PPM and PMU

rouing, 2025-01-03:
Ive noticed that some of components have to use begin and the others (Fuel Guages) have to use init. Now, I am sort of using this library as a reference for my own library but I am having difficulty understanding why begin is exposed to the fuel gauge classes as a member function when you define it.

`    
public XPowersCommon<XPowersAXP202>, public XPowersLibInterface
{
    friend class XPowersCommon<XPowersAXP202>;
`
This works to allow begin to be used...

`
class PowersBQ25896 :
    public XPowersCommon<PowersBQ25896>
{
    friend class XPowersCommon<PowersBQ25896>;
`
Here you have to use init.

I know it doesnt relate to the XPowersLibInterface, so why is it that the begin function is not exposed. I cant see a difference between the 2 device component sources that would cause this.

lewisxhe, 2025-01-03:
There is no difference. Begin calls begin of the XPowersCommon.tpp template class. PowersBQ25896 can also be changed to begin, but the writing method is different.

rouing, 2025-01-03:
Whats interesting is that my IDE's preprocessor says .begin() for the PMU's is valid and of course so is .init(). But what im trying to figure out is why its saying .begin() is not a member of the PowersBQ25896 at all and no matter how you try to call it, it doesnt exist until you remove the `friend class XPowersCommon<PowersBQ25896>` piece. Then it says it exists. 

Now I do know that some IDEs Preprocessors can really screw this up and not get it right, and I havent tested this by running it yet either, but what I would like to find out is why this is the case and if its intended. From what I can tell, it shouldnt matter, the begin() function should be extended into PowersBQ25896. If I use `template <class chipType = PowersBQ25896> friend class XPowersCommon` it works. Although Im not sure if thats achieving the same thing.

lewisxhe, 2025-01-03:
The order of their parameters is different. I use arduino and platformio and they can recognize them normally.
[image]

rouing, 2025-01-03:
[image]
[image]
Its complaining to me about the existence of a member function called begin(); Now If I change the way it handles the friend class, suddenly it works (But it shows as inaccessible without the right arguments)

Meanwhile:

[image]

No complaints here... I have *NO* idea whats different between the PPM and PMU devices, but begin does not get inherited with them and its driving me insane. I know it doesnt matter but I am using this project as reference for my own to have a similar structure and this is really beating me up.

lewisxhe, 2025-01-03:
begin is an internal member and cannot be called from outside. You must give it parameters.

rouing, 2025-01-04:
It really was my IDE's preprocessor freaking out here. I had to rebuild the entire environment but it finally stopped complaining. 

Thanks for your quick response and assistance here. I am new to class factories like this and its been helpful as reference to frame up a HAL for my project using multiple boards, including many Lilygo boards. My goal is to add one extra layer to it to have a namespace that encapsulates all the hardware types.

closed 2025-01-04 by rouing