lewisxhe/XPowersLib #32: [SY6970] NTC Fault IRQ - Improper masking

rouing, 2025-01-05:
POWERS_SY6970_IRQ_NTC_FAULT is defined as the following:
`#define POWERS_SY6970_IRQ_NTC_FAULT(x)                          (bool)(( x & 0xFF ) & 0x03)`

This means a fault status is registered as 0x00000111. However, for this chip, there is no such fault status. 
[image]
The only fault status is combo's of anything but 0x0000000 and 0x00000111

lewisxhe, 2025-01-06:
It only returns the current status of whether the NTC is in error, that is, true or false

If you need to know the connection status, you should use getNTCStatus
https://github.com/lewisxhe/XPowersLib/blob/9f7640bfbc34d8c47e2b2748fbe756e12c482db8/src/PowersSY6970.tpp#L475

rouing, 2025-01-06:
AAAAAAAh thanks!

closed 2025-01-06 by rouing