lewisxhe/XPowersLib #36: [Need advice] getBatteryChargeCurrent() für AXP192?

cyberman54, 2025-03-15:
With the old library it was possible get battery charge and discharge current values from AXP192 (for TTGO T-Beam v1.0).
I can't find out how to get these values with the current library.

lyusupov, 2025-03-15:
https://github.com/lewisxhe/XPowersLib/issues/12

cyberman54, 2025-03-15:
I have AXP192, but my compiler complains:

'class XPowersLibInterface' has no member named 'getBatteryChargeCurrent'; did you mean 'getBatteryPercent'?

cyberman54, 2025-03-15:
I think i'm doing something wrong with integration of the lib.

```
#include "XPowersLib.h"
#define XPOWERS_CHIP_AXP192 1
XPowersPMU pmu;

void PMU_init(void) {
  if (!pmu.begin()) {
        ESP_LOGE(TAG,"No PMU found");
    } else {
        ESP_LOGI(TAG,"PMU type %d found", pmu.getChipModel());
    }

```
results to compiler error

_'bool XPowersCommon<chipType>::begin() [with chipType = XPowersAXP192]' is protected within this context_

cyberman54, 2025-03-16:
solved (i found my errors in my code).

closed 2025-03-16 by cyberman54