lewisxhe/XPowersLib #40: How to control LDOio besides just GPIO0 (like GPIO1, GPIO2, GPIO3.)

yuangezhizao, 2025-05-19:
Hi Lewis,

First off, great work on this library! I'm using it in my study project [ESP-Pocket2](https://github.com/yuangezhizao/ESP-Pocket2).
I've run into a bit of a snag while migrating from the old [AXP202X_Library](https://github.com/lewisxhe/AXP202X_Library) to this new one.
I can't seem to find the **setGPIOMode** and **gpioWrite** functions anymore, which @HwzLoveDz used to use like this:

``` c++
power.setGPIOMode(AXP_GPIO_0, AXP_IO_OUTPUT_HIGH_MODE); // 打开主板指示灯PIO0
power.gpioWrite(AXP_GPIO_0, 1);
```

After digging around a bit, I noticed some functions in the **LDOio** section, like **enableLDOio** and **setLDOioVoltage**.
However, these don't seem to be direct replacements for general GPIO control.

I also found that while **XPOWERS_AXP202_GPIO0_CTL** is used, other GPIO control definitions like **XPOWERS_AXP202_GPIO1_CTL** don't seem to be referenced anywhere in the code

https://github.com/lewisxhe/XPowersLib/blob/master/src/REG/AXP202Constants.h#L55-L60

I'd like to control other GPIOs besides just GPIO0 (like GPIO1, GPIO2, GPIO3.).
Could you point me in the right direction on how to do this? Or perhaps, is support for these other GPIOs planned for a future update?

Thanks!

lewisxhe, 2025-05-20:
哥们 ，和我说中文即可。

这个库是过度AXP192,AXP202 转AXP2101才写的 ，如果你只使用AXP202的话 用之前的旧库就行了 。
我也不会添加这些函数了，因为现在除了AXP2010之外的PMU 全部都停产了，你能买到的也是库存了 。

yuangezhizao, 2025-05-20:
了解啦，多谢。我看`ISSUES`区全是英文还以为都得用英文呢 :)

closed 2025-05-20 by yuangezhizao